NVIDIA-ISAAC-ROS/isaac_ros_common #190: Unable to locate package ros-humble-isaac-ros-centerpose in dev docker container

naslevente, 2025-05-12:
I am receiving an "unable to locate package" error when trying to install the isaac ros centerpose deb package.

Setup:
- Hardware: Jetson Orin Nano Dev Kit
- Software: Jetpack 6.2

I followed the steps to setup the docker container from here: https://nvidia-isaac-ros.github.io/concepts/docker_devenv/index.html#development-environment.

I followed the specific steps for the isaac ros centerpose package as well: https://nvidia-isaac-ros.github.io/repositories_and_packages/isaac_ros_pose_estimation/isaac_ros_centerpose/index.html#quickstart.

Running the run script from the main branch successfully builds the image and starts the container:
```
cd ${ISAAC_ROS_WS}/src/isaac_ros_common && \
./scripts/run_dev.sh
```

Running apt update is successful as well:
```
sudo apt-get update
```

However, when running the following install command for the isaac ros centerpose prebuilt deb package
```
sudo apt-get install -y ros-humble-isaac-ros-centerpose
```
I get the following error:
```
Reading packages lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package ros-humble-isaac-ros-centerpose
```

The isaac ros repository is added to apt. The following line exists in the `/etc/apt/sources.list` file:
```
deb https://isaac.download.nvidia.com/isaac-ros/release-3 jammy release-3.0
```

naslevente, 2025-05-12:
Could be related to the subsequent errors reported in #106 .

naslevente, 2025-05-13:
I noticed that starting the container, the `/etc/apt/sources.list` file contains the `deb https://isaac.download.nvidia.com/isaac-ros/release-3 jammy release-3.0` line. With this however, I was getting the `Unable to locate package` error. Changing that line to `deb https://isaac.download.nvidia.com/isaac-ros/release-3 jammy release-3.0 universe` and running `sudo apt update` fixed the issue for me.

jaiveersinghNV, 2025-05-20:
Thanks for reporting this issue. We think this might have been a temporary blip in the availability of our apt repo. Could you please try again, without any modifications to the sources list?

naslevente, 2025-05-21:
Works fine now without any modifications to the sources list. Thanks!

closed 2025-05-21 by naslevente