NVIDIA-ISAAC-ROS/isaac_ros_common #192: run_dev.sh failing on docker.base build

rob-pointcloud, 2025-05-15:
Hello,

  I just started getting errors when I execute run_dev.

Setup:
Jetson Orin Nano, Jetpack 6.2

This was working fine yesterday, and now I get the error listed below. It seems like this is occuring during the build of "dockerflie.base" and I can see a new commit was just checked in an hour ago:

https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/commit/13bb061e5743a6c43f8ce43619e8051fc5e13d36

It looks like this may be where the docker build is failing for me.

--------------------------------------------------------------------------------------------------

168.0   Preparing metadata (pyproject.toml): finished with status 'error'
168.0   error: subprocess-exited-with-error
168.0   
168.0   × Preparing metadata (pyproject.toml) did not run successfully.
168.0   │ exit code: 1
168.0   ╰─> [30 lines of output]
168.0       Traceback (most recent call last):
168.0         File "/tmp/pip-build-env-lh2bpkar/overlay/local/lib/python3.10/dist-packages/scikit_build_core/_vendor/pyproject_metadata/__init__.py", line 515, in validate
168.0           packaging.utils.canonicalize_name(self.name, validate=True)
168.0       TypeError: canonicalize_name() got an unexpected keyword argument 'validate'
168.0       
168.0       During handling of the above exception, another exception occurred:
168.0       
168.0       Traceback (most recent call last):
168.0         File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 363, in <module>
168.0           main()
168.0         File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 345, in main
168.0           json_out['return_val'] = hook(**hook_input['kwargs'])
168.0         File "/usr/lib/python3/dist-packages/pip/_vendor/pep517/in_process/_in_process.py", line 164, in prepare_metadata_for_build_wheel
168.0           return hook(metadata_directory, config_settings)
168.0         File "/tmp/pip-build-env-lh2bpkar/overlay/local/lib/python3.10/dist-packages/scikit_build_core/build/__init__.py", line 96, in prepare_metadata_for_build_wheel
168.0           return _build_wheel_impl(
168.0         File "/tmp/pip-build-env-lh2bpkar/overlay/local/lib/python3.10/dist-packages/scikit_build_core/build/wheel.py", line 174, in _build_wheel_impl
168.0           return _build_wheel_impl_impl(
168.0         File "/tmp/pip-build-env-lh2bpkar/overlay/local/lib/python3.10/dist-packages/scikit_build_core/build/wheel.py", line 231, in _build_wheel_impl_impl
168.0           metadata = get_standard_metadata(pyproject, settings)
168.0         File "/tmp/pip-build-env-lh2bpkar/overlay/local/lib/python3.10/dist-packages/scikit_build_core/build/metadata.py", line 68, in get_standard_metadata
168.0           metadata = StandardMetadata.from_pyproject(
168.0         File "/tmp/pip-build-env-lh2bpkar/overlay/local/lib/python3.10/dist-packages/scikit_build_core/_vendor/pyproject_metadata/__init__.py", line 428, in from_pyproject
168.0           self = cls(
168.0         File "<string>", line 24, in __init__
168.0         File "/tmp/pip-build-env-lh2bpkar/overlay/local/lib/python3.10/dist-packages/scikit_build_core/_vendor/pyproject_metadata/__init__.py", line 293, in __post_init__
168.0           self.validate()
168.0         File "/tmp/pip-build-env-lh2bpkar/overlay/local/lib/python3.10/dist-packages/scikit_build_core/_vendor/pyproject_metadata/__init__.py", line 516, in validate
168.0           except packaging.utils.InvalidName:
168.0       AttributeError: module 'packaging.utils' has no attribute 'InvalidName'
168.0       [end of output]
168.0   
168.0   note: This error originates from a subprocess, and is likely not a problem with pip.
168.0 error: metadata-generation-failed
168.0 
168.0 × Encountered error while generating package metadata.
168.0 ╰─> See above for output.
168.0 
168.0 note: This is an issue with the package mentioned above, not pip.
168.0 hint: See above for details.
------
Dockerfile.aarch64:155
--------------------
 154 |     # Python3 PIP packages
 155 | >>> RUN python3 -m pip install -U \
 156 | >>>     argcomplete \
 157 | >>>     autopep8 \
 158 | >>>     flake8 \
 159 | >>>     flake8-blind-except \
 160 | >>>     flake8-builtins \
 161 | >>>     flake8-class-newline \
 162 | >>>     flake8-comprehensions \
 163 | >>>     flake8-deprecated \
 164 | >>>     flake8-docstrings \
 165 | >>>     flake8-import-order \
 166 | >>>     flake8-quotes \
 167 | >>>     gpustat==0.6.0 \
 168 | >>>     importlib_resources \
 169 | >>>     networkx \
 170 | >>>     "numpy>=1.24.4,<2" \
 171 | >>>     numpy-quaternion \
 172 | >>>     onnx \
 173 | >>>     pydocstyle \
 174 | >>>     pymongo \
 175 | >>>     pyyaml \
 176 | >>>     "scipy>=1.7.0" \
 177 | >>>     scikit-image \
 178 | >>>     scikit-learn \
 179 | >>>     "setuptools_scm>=6.2" \
 180 | >>>     tqdm \
 181 | >>>     trimesh \
 182 | >>>     "warp-lang>=0.9.0" \
 183 | >>>     "yourdfpy>=0.0.53"
 184 |     
--------------------
ERROR: failed to solve: process "/bin/bash -c python3 -m pip install -U     argcomplete     autopep8     flake8     flake8-blind-except     flake8-builtins     flake8-class-newline     flake8-comprehensions     flake8-deprecated     flake8-docstrings     flake8-import-order     flake8-quotes     gpustat==0.6.0     importlib_resources     networkx     \"numpy>=1.24.4,<2\"     numpy-quaternion     onnx     pydocstyle     pymongo     pyyaml     \"scipy>=1.7.0\"     scikit-image     scikit-learn     \"setuptools_scm>=6.2\"     tqdm     trimesh     \"warp-lang>=0.9.0\"     \"yourdfpy>=0.0.53\"" did not complete successfully: exit code: 1
/ssd/workspaces/pointcloud/src

naslevente, 2025-05-19:
I was running into the same errors. The temporary solution presented in #188 worked for me:

> Also ran into the same issue - changing "RUN python3 -m pip install -U " to "RUN python3 -m pip install --use-deprecated=legacy-resolver -U " solved it for me.

jaiveersinghNV, 2025-05-20:
Thanks for reporting this issue. We believe this has been fixed with our latest Update 8 released late last week. Could you please pull the latest from `main` and try again?
https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/releases/tag/v3.2-8

jaiveersinghNV, 2025-06-06:
Glad to hear that it's working now.

Currently, the interaction between `run_dev.sh`, the `isaac_ros_common` GitHub repo, and our `nvcr.io` container registry is a bit complex:

<details>

<summary>How `run_dev.sh` works</summary>

The `run_dev.sh` script produces a Docker image by stacking a sequence of `Dockerfile.foo` files, allowing different Isaac ROS users to choose which subset of features they require in their development container. To preserve uniqueness, the script appends a hash of the accumulated Dockerfile contents to the end of the composite image tag.

As a convenience to Isaac ROS users, we prebuild a pair of base images and host them on our public NVIDIA container registry `nvcr.io`, so that many users can simply download the existing image instead of having to build it from scratch. By default, the `run_dev.sh` script checks the container registry [here](https://catalog.ngc.nvidia.com/orgs/nvidia/teams/isaac/containers/ros/tags) to see if the user's desired image tag is already available for download.

Since these images are hosted by NVIDIA, we have an obligation to scan for and respond to security vulnerabilities that are discovered in our images. To address these vulnerabilities, we are often required to upgrade dependencies or otherwise make changes to the Dockerfiles in `isaac_ros_common`. The updated Dockerfiles and corresponding refreshed prebuilt images are then released via an [Isaac ROS Update](https://nvidia-isaac-ros.github.io/releases/index.html#isaac-ros-3-2-update-8-may-14-2025), and the old, insecure prebuilt images are removed from our container registry.

Once the Dockerfile contents have changed, `run_dev.sh` generates a new hash, which means the new prebuilt images have different tags than the previous set. As a result, users who have pulled the **latest commit** from `isaac_ros_common` can be sure that they are getting the most secure version of the image, because it has a distinct hash from the previous version. However, the corollary is that anybody using the **previous commit** will notice that `run_dev.sh` can no longer find a prebuilt image; the hash requested by the user's `run_dev.sh` remains the same, but the hashes available on the container registry have changed.
</details>

Additionally, our schedule for building Dockerfiles into Docker images introduces another potential source of confusion:

<details>

<summary>Why rebuilding the same Dockerfiles doesn't always produce the same final image</summary>

As elaborated on in the previous detail section, `run_dev.sh` tags composite images using a hash of the **Dockerfile** contents. Note that this hash is separate from the standard Docker behavior of calculating hashes of **image layer** contents.

Many steps in our Dockerfiles involve downloading resources from the internet, including third-party dependencies, package manager indices, and miscellaneous config files. While we are continually making efforts to improve the reproducibility of these network steps, we currently can't guarantee that a Docker image build started today will fetch the exact same files as a Docker image build started 10 days from now.

In the days leading up to a public Isaac ROS Update, our team builds a fresh image from the soon-to-be-released Dockerfile contents, which is then submitted for publishing to our container registry on the go-live date. Generally, we expect that a user who kicks off a local build of our Dockerfiles shortly after the public update will be able to produce an image that is almost identical (if not exactly identical) to the one we have provided on our container registry.

However, as time continues, the chance that an external resource we depend on gets updated will increase. In some cases, these new dependency versions can suddenly introduce incompatibilities with other dependencies in our image, breaking the build.

Our team does periodically rebuild the public Dockerfiles from scratch in order to check if any dependency issues have occurred. After we notice an issue and have pinpointed exactly what went wrong, we release the fixes as another Isaac ROS Update, using the same process detailed in the previous section.

</details>

Key points:
1. When running on the **latest commit** of `isaac_ros_common`, `run_dev.sh` with default settings should always be able to find the prebuilt base image from NVIDIA's container registry
2. When running on **any older commit** of `isaac_ros_common`, there is **no guarantee** that NVIDIA's container registry will still have the prebuilt image requested by `run_dev.sh`
3. When running on the latest commit of `isaac_ros_common` **soon after that commit was released**, `run_dev.sh` should also be able to build a fresh Docker image locally using the Dockerfile contents from that commit
4. Even when running on the latest commit, if it has been **some time since that commit's release**, a fresh Docker image build **may still fail** if non-NVIDIA third parties have changed some dependency since the original Isaac ROS release

What this means for you:
- **To always have the most secure, up-to-date container:** make sure to always use the latest commit of `isaac_ros_common` and/or fetch the latest image tag from NVIDIA's container registry
    - As per 1, we make every effort to ensure that a prebuilt image corresponding to the latest commit will always be available
[... cut ...] cordingly
    - As per 2, we will remove all insecure images from our public container registry, so we do **not** guarantee that any specific  image will be available forever
    - As per 4, rebuilding a fresh image from the same Dockerfiles does **not** guarantee that you will get the exact same image each time

We are always working to simplify our infrastructure and make it easier for users to choose how they want to use Isaac ROS. We'll have more to share on this subject in a future release.

closed 2025-06-12 by rob-pointcloud