NVIDIA-ISAAC-ROS/isaac_ros_common #197: Unable to build my docker image

javijurado, 2025-06-05:
On my Jenkins pipeline, I'm trying to build my docker image based on both nvidia docker images:

- nvcr.io/nvidia/isaac/ros:x86_64-ros2_humble
- nvcr.io/nvidia/isaac/ros:aarch64-ros2_humble

The issue is failing because the key has expired. These issue has been solved on osrf repository: [link](https://github.com/osrf/docker_images/issues/814)
```bash
#8 35.53 W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://packages.ros.org/ros2-testing/ubuntu focal InRelease: The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
```

I think, you have to update your docker images

jaiveersinghNV, 2025-06-17:
Yes, our latest images should have the new key:
https://catalog.ngc.nvidia.com/orgs/nvidia/teams/isaac/containers/ros/tags

alexhilgart, 2025-06-18:
Hi, it seems that the problem with the new image persists:
```bash
$ docker run -it --rm nvcr.io/nvidia/isaac/ros:x86_64-ros2_humble_23aced29fb80f407b727eec37775e30e bash

closed 2025-07-15 by jaiveersinghNV